FPV-Combat/Main_board_v2 #8: Voltage calibration for OSD broken?

MrMap, 2023-09-28:
Tried the voltage calibration today, no joy.

HW 2.4
FW 2.5.0  

Used PUTTY, serial, 9600, Port 13 (as in com port list)
Board rebooted on FTDI connection but when I attempted to send $ I got no response whatsoever.
Calibration function broken in this FW?

MrMap, 2023-10-06:
Today another reason to try the voltage calibration again, using another FTDi adapter. 
**SUCCESS - but slight change of procedure.**
HW 2.5
FW 2.5.0

Still using PUTTY, now com port 11.
This time I got a response to the $ in the terminal window: "Calib mode:"
No response when sending + or - (and return).
Restarted and entered Calib mode.
Tried pushing various keys and **when pushing "v" (for voltage?) the readings started to stream into the terminal window** and +/- worked for adjustment.

If this little change in procedure was published in documentation, well I missed it.
If not already documented it would be good to do that.

ysoldak, 2024-01-16:
In 2.6.0 firmware you'll be able to calibrate voltage in OSD menu, no serial console needed anymore.
Closing.

closed 2024-01-16 by ysoldak