FPV-Combat/Main_board_v2 #9: Feature request - simple trigger pin for handheld FPVC guns

MrMap, 2023-10-21:
I am thinking of FPVC gaming involving different classes fighting entities, e.g. planes and handheld guns.
For the handheld gun it would be convenient to have the trigger controlled by a simple mentary trigger switch on the gun - no need for PWM there.

I can think of two solutions:
1. Configurable in board settings, trigger: PWM / Simple switch
2. Trigger port auto detection: Recognize PWM / activate trigger when input pulled down for > 10 ms

ysoldak, 2024-01-16:
Good idea, thanks!

MrMap, 2024-02-27:
I poke at this issue to make some noise.
I think this feature would be really simple to implement - compared to the other advanced stuff you do?

The feature would make it much more simple to assemble handheld guns - good usage of the 2.5 boards that will soon be surplus.

ysoldak, 2024-03-05:
Not that straightforward to implement, unfortunately.
Also if we do not auto-detect, we need to make it configurable via menu, stored in persistent storage, small things like that, but together can result in quite some code to implement, test and maintain.
I'd first want to explore possibility to implement auto-detect for that, as you suggested in option 2.