FPV-Combat/Main_board_v2 #12: Shooting/beeing hit Audio feedback

chaos5235, 2024-01-11:
Can we get a beep or chirp when we are shooting or get shot?

e.g. 
- 1 chirp for shot 
- 2 chirps for beeing hit

ysoldak, 2024-01-16:
Good idea, thanks!

ysoldak, 2024-01-29:
I think we do have some (different) sound on such events already, please confirm @chaos5235

chaos5235, 2024-01-29:
I don't hear beeping from the when I shoot ?!

chaos5235, 2024-01-30:
I hear sound when I get hit - but not when I shoot!

chaos5235, 2024-02-08:
maybe something good distinguishable . . . on getting hit we have the 3x beep..... maybe for shooting just a really short beep would be cool

ioteo, 2024-02-20:
We could consider to add an option for the working mode of the buzzer eventually...
DISABLED: the buzzer doesn't play ANY sound not even at booting time
NORMAL: different "beeps" at boot, arming, hit received
FULL: same as normal + menu navigation, and during shooting burst...
What do you think @chaos5235 and @ysoldak ?

ysoldak, 2024-02-21:
@ioteo  Good idea, let's see what we can do!
You know how to put the buzzer in different states, hardware wise?
And I suppose it only be possible on HW 2.6?