FPV-Combat/Main_board_v2 #22: Hardware: 2.6.0. - bigger Pads and Holes for PowerConnection

chaos5235, 2024-01-18:
The actual Holes and solder area for Supply Voltage is pretty small  no normal power cable can be pushed through the holes.

It would be great to have an additional big solder pad to enable users to use a descent power connection cable alternativly to servo connector pins

 
[image]

ioteo, 2024-02-20:
@chaos5235 so pads without holes or those pads in addition to the existing ones? Since there is not really much space there...
[image]

chaos5235, 2024-02-23:
maybe just 2 bigger holes instad of 2 small ones....

ioteo, 2024-03-08:
Completed with V2.6_r0.5, maximized pad's size and increased also holes diameter to 1.5mm without changing the shape or position maintaining same pitch as the rest of the board (2.54mm).

closed 2024-03-08 by ioteo