FPV-Combat/Main_board_v2 #30: HDZero VTX management - requirements

MrMap, 2024-01-30:
In order to use HDZero video system with the new 2.6 digital MSP Canvas feature there needs to be an easy and robust way of controlling the HDZ VTX.

### Background:
Unlike DJI and WS, **HD Zero VTX management relies heavily on being hooked up to a serial data channel, normally achieved using Flight Controller UART**

DJI and WS goggles feature full range telemetry & VTX control transmission sent from the goggles themselves and this is why the DJI / WS VTX are controlled from the goggles menu - thus the system is self contained.

The HDZero is basically a simplex system - the goggles only receive incoming video stream and _it does not transmit_ (see footnote) - hence the goggles cannot control the VTX (band, channel, power). AFAIK this is a design choice to avoid pilot zone 5 GHz band spectral pollution during multi pilot activities - like racing. As for the HD Zero Freestyle (v.1 & 2) VTX there are three methods to manipulate the VTX settings:

1. Using the VTX MSP serial port with MSP VTX protocol (usually by running LUA on EdgeTX hand unit - realayed to VTX via Flight Controller). This is easy, robust and offers smart features. Using the ELRS Backpack feature, the goggle  can be made to automatically follow the VTX channel. The FC OSD will display VTX band, channel and power.
2. Using the VTX **Smart Audio** port (TBS SA). Until MSP VTX made it into INAV and Betaflight (not so long ago), SA was the only "workable" way to control the VTX. SA required a dedicated FC UART TX channel to be connected to the VTX SA port, basically offering same functionality as the above MSP VTX - but slower and _way less robust IMHO_.
3. Using the dedicated keyboard, physically plugged into the VTX. Up until today I did not figure out how to properly use this method and anyway it´s not something you would want to fiddle with in field (or at all). Physically tiny and delicate connector and not an intuitive way of having the settings confirmed.

As long as there is a Flight Controller in the loop - methods 1 and 2 are available for VTX control.
But if we unhook the HDZ VTX from the FC - to hook the VTX up to FPV Combat board MSP port instead - we loose the MSP VTX control. 
The Smart Audio could be a fallback alternative (using the dedicated FC UART and VTX SA input) - though without OSD info on VTX settings. Reverting from MSP VTX to SA may be percieved as... painful.

### Solution request, short term
Provide means to send MSP VTX data from the FPVC 2.6 MSP port to the VTX MSP port.
The VTX settings parameters would have to be entered into the 2.6 board first, maybe using the Gadget (radio port) or via USB interface.

### Solution request, long term
Merge the MSP data from Flight Controller and combat board before reaching the VTX.
I know this is already on the developers roadmap.

Footnote: The HD Zero goggles can actually transmit - using the ELRS Backpack feature. The Backpack is a short range ESP32 based system using ESP-NOW 2.4 GHz protocol - enabling for smart features. It does not interfere with 5GHz band video transmissions.
One application example being HDZ goggles following VTX video channel commands sent from RC handset LUA. 
Other example, HDZ goggles built-in Head Tracker output (wireless) injected into ELRS RC channels for camera gimbal control.

ysoldak, 2024-01-31:
Thanks!

Shall investigate if we must have this in Combat firmware or can solve this with a helper MSP-merger board.

MrMap, 2024-01-31:
> Thanks!
> 
> Shall investigate if we must have this in Combat firmware or can solve this with a helper MSP-merger board.

Heads Up regarding HDZ combability. 
It´s is possible that this (MSP helper board) needs to be coordinated with the HDZero VTX firmware development.
As I recall, me being among the early testers of MSP VTX on INAV to HDZ VTX, while MSP VTX was up and running for many months on Betaflight-> HDZ VTX, corresponding functionality on INAV came later and involved releases in both INAV and HDZ VTX firmware. I guess it will show once you start digging into that...

ysoldak, 2024-02-21:
Shall be possible with new code in 2.6.1 that enables MSP pass through functionality.

ysoldak, 2024-02-25:
Moved to 2.7.0, MSP pass through needs more testing.

ysoldak, 2024-05-14:
@MrMap please confirm MSP passthrough works fine (on 2.7.0 beta) for your application.

MrMap, 2024-05-15:
MSP passthrough worked normally when tested today with 2.7.0 B5
INAV 7.1 and HDZero video system