FPV-Combat/Main_board_v2 #34: A function to check if HC12 is conncted and operable

chaos5235, 2024-01-31:
It would be nice to have a veryfication in the Setup menue to see if HC12 is connected and if its working

ysoldak, 2024-01-31:
How you imagine this?
The test station can help you with such test, but I don't see how to implement this having single HC-12 board installed on model.
Also can use The Gadget to try and communicate with the model.

ysoldak, 2024-02-08:
We can try and use DTR pin of the board to put HC12 in "SET" mode and see if board replies "AT" commands.

ioteo, 2024-02-20:
@ysoldak good idea, however DTR pin can't be used from MCU as GPIO for multiple reasons...
it could be driven LOW by the serial adapter but that means that the HC-12 test can be performed only through USB and some scripts.
In addition, checking AT commands will not ensure that the transceiver is actually fully working (especially the RF).

ioteo, 2024-02-20:
@ysoldak @chaos5235 I honestly think that is not really doable or doesn't worth it.
Alternative devices like "the gadget" and the static target already provide a way better solution to check the functionality.
The only real simple and doable implementation in my opinion will always require an other module configured as "loopback" (TX and RX connected together and module powered via 5V, power bank or single LiPo cell).
Then the test could simply be performed sending data over serial and waiting for exact same data coming back a few milliseconds later...
However also this approach doesn't comes free of downsides and other limitations.

ysoldak, 2024-02-21:
Not really doable. Closing for now as "won't fix". May return to this later if we find a way.

closed 2024-02-21 by ysoldak