FPV-Combat/Main_board_v2 #38: Status LED should be ON (or blinking) during startup

ioteo, 2024-02-20:
When powering the system the STATUS led after the first blink should stay "ON" or blinking during the whole startup procedure indicating that the system is powered and booting. Otherwise the longer boot time might be misinterpreted as "bricked" board.

ysoldak, 2024-02-21:
OK, can have that: blue LED is ON or blinking whole time during boot. This shall work nice on HW 2.6.

On HW 2.5, we have blue LED connected to buzzer. What shall we do where?
Blink red LED, blink blue and buzz or do nothing?

ioteo, 2024-02-22:
On V2.5 the buzzer is ON when STATUS led is turned OFF, so multiple options:
1) STATUS led is ON during the boot time
2) STATUS led is ON and quick "BEEP" is generated every second
3) blink the HIT led instead
For me any of them sounds good, probably I like the most second or first option

ysoldak, 2024-02-25:
Fixed, implemented option 1 in 2.6.1

closed 2024-02-25 by ysoldak