FPV-Combat/Main_board_v2 #41: Trigger PWM input port malfunctions at high PWM refresh rates (ver. 2.6.1-B2)

MrMap, 2024-02-22:
version 2.6.1-B2

I have implemented the (Trigger channel) menu navigation logic in both my TX16S radios - assigning 1700us respective 1850us to momentary buttons. Menu navigation worked pretty well using control inputs from a FrSky S8R receiver.
Then I replicated the setup in my other radio that I use for RaptuS with Flight Controllers. Did not work well; 1700us input produced desired menu response, but 1850us input produced the same response as 1700us. Like if the trigger input port could not distinguish between 1700 and 1850. The acual PWM value was verified on a ISDT universal checker - displaying pulse width and refresh rate.
How could this be? What was the difference between the working setup and the faulty one?

Well, the S8R outputs 50Hz servo refresh rate. The flight controller as configured produced 330Hz.
As I successively stepped down the (INAV) FC servo refresh rate, the combat board begun to respond as supposed to. 
At 100Hz refresh rate it mostly worked, though not 100%.
Lowering to 50Hz produced pretty much same result as when using the S8R RX. (still not 100%, but usable).

Conclusion: 
As for now, the combat board v 2.6 trigger PWM input port is unable to reliably cope with refresh rates higher than 50Hz.
Using v 2.5 hardware and software on 330Hz was never a problem. Though there was less pulse values to distinguish between, basically low and high. Now, with FW 2.6 there is also 1700 and 1850us.

Request: Fix trigger port capability to cope with high PWM refresh rates as commonly used with Flight Controllers.
Hight servo refresh rate is one of the factors that enable the FC PIFF loop to produce that stable, "locked in" flight response many pilots desire. Featuring crisp aircraft responses to stick inputs and cutting through turbulent weather "on rails". Reverting to 50Hz will of course still be flyable, but if at all avoidable, the combat board should not impose PWM refresh rate restricions in the same way as the old analog servos do. It would be a step backwards and not flattering for the Combat system as such - indicating low performance hardware.

Attached two screen shots of post in Facebook INAV Fixed Wing Group, on the topic of servo refresh rate:

[image]
[image]

ysoldak, 2024-02-25:
Great explanation for usability of higher refresh rates, thanks, @MrMap !
Hopefully, we can have it fixed, see #44

ysoldak, 2024-02-25:
Fixed in 2.6.1

closed 2024-02-25 by ysoldak