FPV-Combat/Main_board_v2 #47: Consideration - FPVC OSD crosshair ON/OFF also toggles trace fire animation. I suggest it should not.

MrMap, 2024-04-02:
**In an (slightly rushed) attempt to better align my OSD crosshair with the IR-gun light beams** I decided to disable the FPVC crosshairs and enable the INAV crosshair. 
Why?  Because the FPVC crosshair is locked in position (at least for now) while the INAV crosshair can be adjusted up/down +2, +1, 0 , -1, -2.   
**Adjusting the crosshair might often be a lot more convenient than attempting to physically re-align the IR-guns.**

This workaround was successful but during the Eastern meetup I kept wondering why I did not see any OSD trace fire when shooting.
After some confused testing today I finally realized what happened to the trace fire. It was disabled together with the FPVC crosshair.

I can see some logic in this but as long as the FPVC crosshair position is not available for calibration / position adjustment, I suggest that FPVC OSD crosshair type selection (no crosshair) only affects the crosshair and leaves the trace fire on.

If somebody would like to disable both the crosshairs and the trace fire I suppose they could just use the "HIDE sight and traces" radio setup for the trigger channel?

ysoldak, 2024-05-14:
We may just want to add a setting: "Show traces ON/OFF"

chaos5235, 2024-07-18:
yeah you have additionally the firing indication at the bottom of the screen to verify that you r shooting