FPV-Combat/Main_board_v2 #48: HW 2.4/5  OSD only visible in NTSC mode - PAL not

MrMap, 2024-04-03:
HW 2.4 or 2.5
FW 2.6.3
Analog camera Caddx Ratel 2

Combat system OSD only visible when camera operated in NTSC mode.
In PAL mode the video image works as usual, but no OSD.

When system is started without camera connected, OSD is visible.
As soon as camera is plugged in (in PAL mode) OSD dissappears.
Confirmed on two combat boards and two cameras.

MrMap, 2024-04-06:
Same problem encountered using Foxeer Falkor camera.
HW 2.5
FW 2.6.3

If the user cannot change from PAL to NTSC he will not be able to enjoy the OSD (at least not when the camera is connected).
So this may be a pretty high priority?

ioteo, 2024-07-18:
with current version V2.8 video mode detection still seems to have some problems.
I suggest to restore the original approach that was checking video mode in main loop or every reset of the board so the user can change camera's video mode and the system will automatically detect without any additional configuration.
As tested during the camp, PAL seems to not be working correctly.