FPV-Combat/Main_board_v2 #49: Is it possible to have OSD with DJI  AirUnit o3 and Googles 2?

feffe81, 2024-05-06:


ysoldak, 2024-05-14:
Yes, DJI O3 Air Unit works.
Please flash latest firmware, it is 2.7.0-B5 now.

See firmware flash instructions (a bit low level, we'll have Configurator/Flasher UI soon)
https://github.com/FPV-Combat/Main_board_v2/wiki/Flash-Firmware

feffe81, 2024-05-14:
Nice. I have not found any documentation how to wire it up. Is there something available?

Gruß 
Steffen

ysoldak, 2024-05-15:
You connect O3 and combat board same as you would connect it to a flight controller.

On the combat board (hardware version 2.6, latest) there are MSP pins.
Connect O3 RX to combat board MSP TX and O3 TX to combat board MSP RX.

feffe81, 2024-05-15:
Thx

Gruß 
Steffen

feffe81, 2024-05-16:
Is there any documentation or source code available for MSP with O3? I would also like to include it in my project on GitHub

ysoldak, 2024-05-16:
> Is there any documentation or source code available for MSP with O3? I would also like to include it in my project on GitHub

No, sorry, no our code available.
But look here, you may find this useful
https://github.com/ramiss/arduino_DJI_03_RC_ARM